KonradIT/goprowifihack #152: hero5black  remains in pairing mode

habakuk39, 2019-05-13:
@konradIT:

I am on Ubuntu 18.04 on ondroidC2 with GoProHero5black:

1.) I can pair (bluettoothctl) and connect (gatttool) correctly without any problems!
2.) I can connect to my GpH5b wifi correctly without any problems (wpa2psk, 2,4 MHz)

But the cameradoesn't seem to accept this connectivity and NEVER leaves the paring mode! So it finally ends up in breaking up the pairing attempt and killing the wifi AP.

No matter what I try, no successfull paring on the camera side....

KonradIT, 2019-05-13:
It never leaves the pairing mode because you need to execute the following URL: http://10.5.5.9/gp/gpControl/command/wireless/pair/complete?success=1&deviceName=DESKTOP

habakuk39, 2019-05-13:
great thanks so much

marcotidei, 2024-02-16:
Same problem but I don't understand the solution.
After 8 hours of searching for the solution, I think it's time to ask for help.

I want to connect my Hero 5 Black to my Raspberry Pi Zero W. I cannot have the camera completing the pairing.
I intend to use Bluetooth to wake up the camera and shoot photos, I'm not planning to use the Wi-Fi at all.
How can I execute the command posted above?

This is the order of my commands:

```
advertise on
trust XX.XX.XX.XX
pair XX.XX.XX.XX
connect XX.XX.XX.XX
```

Then the prompt on my Pi shows [GoPro 6092] but the camera remains in pairing mode.

Thanks.