KonradIT/goprowifihack #153: how to monitor the battery status on hero7 black?

anandkoirala, 2019-05-23:
**Problem:**
could you please share the HTTP request command for checking the battery status on hero 7...
thank you
Describe your problem as accurately as possible, include key details that help resolve your issue.

**Details:**

* GoPro Camera(s):
* Firmware Version:
* Steps to reproduce:
* Happens every time? [Y/N]:

RobsyRocket, 2019-05-23:
You could try to query the details via the status API endpoint mentioned in the [documentation of the HERO4](https://github.com/KonradIT/goprowifihack/blob/master/HERO4/CameraStatus.md). I assume that it is still valid because it is referenced in the main [HERO7 documentation](https://github.com/KonradIT/goprowifihack/blob/master/HERO7/README.md).

KonradIT, 2019-05-23:
Battery status ID is the same as H4 yes.

anandkoirala, 2019-05-24:
thanks @RobsyRocket and @KonradIT