KonradIT/goprowifihack #154: v4into doesn't return JSON

ram-sh, 2019-05-24:
Hello,

when using the "videoinfo" the camera returns the expected JSON:

http://10.5.5.9:8080/gp/gpMediaMetadata?p=100GOPRO/GOPR2410.MP4&t=videoinfo
{"dur":"23","avc_profile":"100","profile":"42","tag_count":1,"tags":[9459]}

however when calling the same with t=v4info like so:
http://10.5.5.9:8080/gp/gpMediaMetadata?p=100GOPRO/GOPR2410.MP4&t=v4info 
the camera doesn't return a JSON but a thumbnail image.

* GoPro Camera(s): HERO 4 Black
* Firmware Version: v05.00
* Steps to reproduce: as described above
* Happens every time? [Y/N]: Y

Many thanks in advance!!!

KonradIT, 2019-05-24:
Only for photos.

ram-sh, 2019-05-24:
that explains that :) 

are there other query strings besides "videoinfo" that can be used? I mainly interested to get the video rez 

in any case, many thanks for your awesome work and support!

KonradIT, 2019-05-25:
(only for photos - in HERO4 Black. HERO5 Black and newer support the feature)

ram-sh, 2019-05-25:
got it, many thanks!

closed 2019-06-15 by ram-sh