KonradIT/goprowifihack #157: how to turn on wifi on gopro5 black?

victory93, 2019-06-24:
**Problem:**

Describe your problem as accurately as possible, include key details that help resolve your issue.

**Details:**

* GoPro Camera(s):  hero5 black
* Firmware Version:
* Steps to reproduce:
* Happens every time? [Y/N]: 

i use virtual server, That server is Unable to set wol.

Is there any other way to turn the wifi on?

Is there any way to solve the problem?

please Let me know if you have a solution. thank you

KonradIT, 2019-06-24:
Turn the wifi on? WoL wakes up the camera, you can turn on the wifi via Bluetooth. If you mean keep the camera alive you can use TCP packets or start streaming.

victory93, 2019-06-25:
how to send wake on lan command?

If I don't use Wi-Fi, can I set up wol?

closed 2019-06-26 by victory93