KonradIT/goprowifihack #162: HERO 8 Black Live Stream

linkevinlin1, 2019-10-31:
**Problem:**

Describe your problem as accurately as possible, include key details that help resolve your issue.
I've tried to use the live stream instruction for HERO 7. This command (http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart) gives the following error:

```
{
"version": "6.00",
"path_info": "execute",
"query_string": "p1=gpStream&a1=proto_v2&c1=restart",
"error_code":-3,
"error_msg": "500 Internal Server Error
",
"function": "gpcontrol_cgi_handler",
"line": 190
}
```

Is it possible to get udp live stream on to PC? 

Thank you. 

**Details:**

* GoPro Camera(s): HERO 8 black
* Firmware Version: 01.20
* Steps to reproduce:
* Happens every time? [Y/N]: Y

KonradIT, 2019-11-02:
Will fix this tomorrow.

linkevinlin1, 2019-11-04:
Thanks for the reply. Any update on this?

fleather2, 2019-11-05:
Thanks for the fix, the command now executes successfully with status: "0". However, the url given in your original tutorial ,udp://10.5.5.9:8554, is not working. I am trying to open the stream in VLC media player. Any advice? Would you rather I open a new issue than continue this thread?