KonradIT/goprowifihack #163: Connect to Hero 8 over WiFi without Bluetooth?

demiantres, 2019-11-02:
**Problem:**

I can see the Hero 8 WiFi only when the camera is connected with the Android GoPro app. Is it possible to enable the WiFi without using the GoPro app so that I can connect my PC to the camera?

**Details:**

* GoPro Camera(s): Hero 8
* Firmware Version: 1.2
* Steps to reproduce:
* Happens every time? [Y/N]:

KonradIT, 2019-11-02:
Looking into this as I can repro this on my MAX.

demiantres, 2019-11-03:
This works:
1.) Switch the Hero 8 to App pairing mode.
2.) Connect to the Hero 8 with Bluetooth. Now we can connect to the WiFi.

The challenge is that the Hero 8 pairing flow has changed somehow - I cannot get past the Hero 8 pairing screen.

demiantres, 2019-11-09:
Do you have more details about authentication?

KonradIT, 2019-11-09:
Use `hcitool`:

hciconfig auth
hciconfig encrypt

demiantres, 2019-11-09:
Do you need the gpControl json file from the Hero 8?

KonradIT, 2019-11-09:
Sure, send it over (make sure to redact your MAC address at the end!)

demiantres, 2019-11-09:
`{"version":2,"schema_version":5,"modes":[{"id":12,"display_name":"Video"},{"id":15,"display_name":"Looping"},{"id":16,"display_name":"Single Photo"},{"id":17,"display_name":"Photo"},{"id":18,"display_name":"Night Photo"},{"id":19,"display_name":"Burst Photo"},{"id":13,"display_name":"Time Lapse Video"},{"id":20,"display_name":"Time Lapse Photo"},{"id":21,"display_name":"Night Lapse Photo"},{"id":24,"display_name":"Time Warp Video"},{"id":25,"display_name":"Live Burst"},{"id":26,"display_name":"Night Lapse Video"},{"id":27,"display_name":"Slo-Mo"},{"id":4,"display_name":"Playback"},{"id":5,"display_name":"Setup"},{"id":22,"display_name":"Broadcast Record"},{"id":23,"display_name":"Broadcast"}],"ui_mode_groups":[{"display_name":"VIDEO","id":1000,"modes":[12,15,27]},{"display_name":"PHOTO","id":1001,"modes":[17,18,19,25]},{"display_name":"TIME LAPSE","id":1002,"modes":[24,13,20,21,26]}],"settings":[{"display_name":"Resolution","id":2,"options":[{"id":1,"display_name":"4K","value":1},{"id":18,"display_name":"4K 4:3","value":18},{"id":4,"display_name":"2.7K","value":4},{"id":6,"display_name":"2.7K 4:3","value":6},{"id":7,"display_name":"1440","value":7},{"id":9,"display_name":"1080","value":9}]},{"display_name":"Frames Per Second","id":3,"options":[{"id":0,"display_name":"240","value":0},{"id":13,"display_name":"200","value":13},{"id":1,"display_name":"120","value":1},{"id":2,"display_name":"100","value":2},{"id":5,"display_name":"60","value":5},{"id":6,"display_name":"50","value":6},{"id":8,"display_name":"30","value":8},{"id":9,"display_name":"25","value":9},{"id":10,"display_name":"24","value":10}]},{"display_name":"Interval","id":5,"options":[{"id":10,"display_name":"60 Minutes","value":10},{"id":9,"display_name":"30 Minutes","value":9},{"id":8,"display_name":"5 Minutes","value":8},{"id":7,"display_name":"2 Minutes","value":7},{"id":6,"display_name":"60 Seconds","value":6},{"id":5,"display_name":"30 Seconds","value":5},{"id":4,"display_name":"10 Seconds","value":4},{"id":3,"display_name":"5 Seconds","value":3},{"id":2,"display_name":"2 Seconds","value":2},{"id":1,"display_name":"1 Second","value":1},{"id":0,"display_name":"0.5 Seconds","value":0}]},{"display_name":"Interval","id":6,"options":[{"id":0,"display_name":"Max","value":0},{"id":4,"display_name":"120 Minutes","value":4},{"id":3,"display_name":"60 Minutes","value":3},{"id":2,"display_name":"20 Minutes","value":2},{"id":1,"display_name":"5 Minutes","value":1}]},{"display_name":"Low Light","id":8,"options":[{"id":1,"display_name":"ON","value":1},{"id":0,"display_name":"OFF","value":0}]},{"display_name":"ISO MAX","id":13,"options":[{"id":0,"display_name":"6400","value":0},{"id":3,"display_name":"3200","value":3},{"id":1,"display_name":"1600","value":1},{"id":5,"display_name":"1200","value":5},{"id":6,"display_name":"1000","value":6},{"id":4,"display_name":"800","value":4},{"id":2,"display_name":"400","value":2},{"id":7,"display_name":"200","value":7},{"id":8,"display_name":"100","value":8}]},{"display_name":"Shutter","id":19,"options":[{"id":6,"display_name":"30 Seconds","value":6},{"id":5,"display_name":"20 Seconds","value":5},{"id":4,"display_name":"15 Seconds","value":4},{"id":3,"display_name":"10 Seconds","value":3},{"id":2,"display_name":"5 Seconds","value":2},{"id":1,"display_name":"2 Seconds","value":1},{"id":0,"display_name":"Auto","value":0}]},{"display_name":"ISO MAX","id":24,"options":[{"id":5,"display_name":"3200","value":5},{"id":4,"display_name":"1600","value":4},{"id":0,"display_name":"800","value":0},{"id":1,"display_name":"400","value":1},{"id":2,"display_name":"200","value":2},{"id":3,"display_name":"100","value":3}]},{"display_name":"Interval","id":30,"options":[{"id":100,"display_name":"60 Minutes","value":3600},{"id":101,"display_name":"30 Minutes","value":1800},{"id":102,"display_name":"5 Minutes","value":300},{"id":103,"display_name":"2 Minutes","value":120},{"id":104,"display_name":"60 Seconds","value":60},{"id":105,"display_name":"30 Seconds","value":30},{"id":106,"display_name":"10 Seconds","value":10},{"id":107,"display_name":"5 Seconds","value":5},{"id":108,"display_name":"2 Seconds","value":2},{"id":109,"display_name":"1 Second","value":1},{"id":110,"display_name":"0.5 Seconds","value":0}]},{"display_name":"Shutter","id":31,"options":[{"id":6,"display_name":"30 Seconds","value":6},{"id":5,"display_name":"20 Seconds","value":5},{"id":4,"display_name":"15 Seconds","value":4},{"id":3,"display_name":"10 Seconds","value":3},{"id":2,"display_name":"5 Seconds","value":2},{"id":1,"display_name":"2 Seconds","value":1},{"id":0,"display_name":"Auto","value":0}]},{"display_name":"Interval","id":32,"options":[{"id":3600,"display_name":"60 Minutes","value":3600},{"id":1800,"display_name":"30 Minutes","value":1800},{"id":300,"display_name":"5 Minutes","value":300},{"id":120,"display_name":"2 Minutes","value":120},{"id":60,"display_name":"1 Minute","value":60},{"id":30,"display_name":"30 Seconds","value":30},{"id":20,"display_name":"20 Seconds","value":20},{"id":15,"display_name":"15 Seconds","value":15},{"id":10,"display_name":"10 Seconds","value":10},{"id":5,"display_name":"5 Seconds","value":5},{"id":4,"display_name":"4 Seconds","value":4},{"id":3601,"display_name":"Auto","value":3601}]},{"display_name":"ISO MAX","id":37,"options":[{"id":5,"display_name":"3200","value":5},{"id":4,"display_name":"1600","value":4},{"id":0,"display_name":"800","value":0},{"id":1,"display_name":"400","value":1},{"id":2,"display_name":"200","value" [... cut ...] PCAMERA_BATTERY_LEVEL_ID","display_name":"Battery Level","widget_type":"readonly","precedence":1},{"command_key":"GPCAMERA_SDCARD_CAPACITY_ID","display_name":"SD Card Capacity","widget_type":"child","precedence":1}]}],"info":{"model_number":50,"model_name":"HERO8 Black","firmware_version":"HD8.01.01.20.00","serial_number":"XXXXX","board_type":"0x05","ap_mac":"XXXXX","ap_ssid":"XXXXX","ap_has_default_credentials":"0","git_sha1":"XXXXX","capabilities":"16","lens_count":"1","update_required":"0"}}`

demiantres, 2019-12-12:
Have you made progress with the Bluetooth auth? I went through the camera logs (from https://github.com/hypoxic/GoProHERO8) and recorded my own BT HCI logs but without real progress. The hypoxic logs show a device and app PIN code (ble_pin=1725 app_pin=6850) and my BT logs have some read/write interaction but there is no clear pattern.

whc2001, 2019-12-12:
Just got a HERO 8, tried to capture the HCI log but unable to understand it. Here is the log for further analysing. (MAC address has been replaced, host is AA:BB:CC:DD:EE:FF and camera is 11:22:33:44:55:66)

[btsnoop_hci_nomac.log](https://github.com/KonradIT/goprowifihack/files/3957418/btsnoop_hci_nomac.log)

demiantres, 2019-12-13:
Have you included the pairing process in the log (first camera connect with your phone)?

demiantres, 2019-12-13:
Here are some further observations using a custom connection with the Android Bluetooth API:

**Case 1:** The GoPro network settings have been reset to **default** (factory):

1. Connection to the GoPro with the Android Bluetooth API works. But on some phones the WiFi SSID and password are not transmitted (failure: "Insufficient Authentication" or "Read not permitted"). I can read other stuff like battery level, model string (HD8...) and other common properties.
2. Writing **Bluetooth commands**, such as start/stop recording or turn WiFi on **do not work**.

**Case 2:** The GoPro has been paired with the Android device using the **GoPro app**:

1. Connection to the GoPro over Bluetooth with the Android API works. WiFi SSID and password are transmitted.
2. **Bluetooth commands** (start/stop recording, WiFi on/off) **work**.

So it seems as if the GoPro app does some kind of registration on the GoPro camera. This registration must involve something unique because after the initial pairing with the GoPro app I can use the standard Bluetooth API for everything. The unique token could be the MAC address but I do not think so because we do not have access to the MAC address on iOS.

whc2001, 2019-12-13:
> Have you included the pairing process in the log (first camera connect with your phone)?

The capture includes the packets from these procedures:
 - Turn on bluetooth
 - Open GoPro app (data cleared)
 - Click "Find device"
 - Click "Connect device"
 - When the device name prompt appears, click not now, then close the app and bluetooth.

Also the GoPro has been paired with my main phone before. Tell me if you need a capture after factory reset.

shubham9436, 2021-01-04:
Any progress on how Connect to Hero 8 over WiFi without Bluetooth?