KonradIT/goprowifihack #164: Hero 8 Stream Over Wifi

fleather2, 2019-11-07:
**Problem:**

I am attempting to stream the Hero 8 over Wifi to a PC using the procedure documented for the Hero 7 and following the advice you gave in this thread: https://github.com/KonradIT/goprowifihack/issues/162
I have not been able to successfully view a stream on the PC.

**Details:**
I am using the GoPro Hero 8, a windows laptop, a linux Mint laptop, and my Samsung Galaxy S10 with the GoPro app installed.

* GoPro Camera(s): Hero 8 black
* Firmware Version: 01.20

* Steps to reproduce:

My current procedure uses the phone app, the linux laptop, and the gopro. Please excuse my limited knowledge of network technology.

1) Connect to GoPro via the GoPro app on your cell phone. This seems to "prime" the GoPro's hotspot and without doing this, the network may not appear
2) On the computer, wait for the GoPro's network to appear (it will be called whatever the GoPro is called in the app, and it may be necessary to disconnect from your current network), and connect to it. The password for the network can be found on the GoPro by swiping down -> Preferences -> Connections -> Camera info. Once connected to the network, it is not necessary to have the app open anymore. 
3) Open VLC Player and use Ctrl + N to view a stream over a network. In the network URL box, type "udp://@:8554" and press play.
I have also tried udp://@10.5.5.101:8554 and udp://@10.5.5.9:8554 with inconsistent success.
4) Send a GET request to the GoPro with the following URL: "10.5.5.9/gp/gpControl/execute?p1=gpStream&c1=restart" This can be done by simply typing the URL into a browser or by using a different method, such as the Python requests library. If your app is still open, you will notice that this freezes the stream until you reconnect to the GoPro.
5) Repeat steps 3 & 4 at random and, if you are lucky, you _might_ get one corrupted-looking frame of a stream. Try mixing up steps 3 & 4. 

* Happens every time? [Y/N]: N

This works sometimes on the linux laptop, and it hasn't yet worked on the windows laptop.
I am curious as to how you found out about the :8554 port and the GET URL. I'm no web dev and I have no idea how to interpret the JSON info. Perhaps they changed this for the new cameras?

Thanks so much for your help.

KonradIT, 2019-11-07:
I still need to get a H8 for debugging, waiting for confirmation that GoPro will supply me with one. A few pointers: use MPV, not VLC. Start the WiFi using phone app and then close phone app, then connect to WiFi, start keep alive thread, then start stream with URL from API (new URI, sorry everyone). Then the UDP stream will be at 8554. Tested on my GoPro MAX with no problem, same generation as H8.

fleather2, 2019-11-12:
Excellent news! I've gotten the Hero 8 to stream! The fix was frustratingly simple, but perhaps we could work together to streamline the procedure.
I open up one terminal with python 3 and do the following:
`from gprocam import GoProCamera, constants`
`gpc = GoProCamera.GoPro()`
`gpc.stream("udp:127.0.0.1:10000")`
Then I open a browser and use the url discussed in [https://github.com/KonradIT/goprowifihack/issues/162](url): 10.5.5.9/gp/gpControl/execute?p1=gpStream&c1=restart
Then, I will wait until the speed parameter in the terminal is below speed=1.10x, and I use ffplay to view the stream:
`ffplay -probesize 32 -sync video -an -sn -fast -infbuf -f mpegts -i udp://127.0.0.1:10000`
The stream will show up with less than 2 seconds delay. I ended up using the Linux machine, and I haven't tried it on windows yet. 
I tried editing the url in the livestream() function in GoProCamera.py to the new url (line 1141), but it made no difference, I still have to send the request after the stream has started. 
Furthermore, I'm curious if you have ever tried to view the stream in GStreamer and if you've had any success. Thanks again for your help.

fleather2, 2019-12-03:
Wanted to close the issue and leave an update for anyone who reads this thread, the steps necessary to stream for the GoPro Hero 8 on windows are as follows (as it worked for me):

Connect to the GoPro's wifi network by first connecting to it via your phone app, then connecting to it after it appears on your list of networks. 
Using konrad's [gopro-py-api](https://github.com/KonradIT/gopro-py-api), connect to the GoPro and keep it alive with the following python3 commands:
`from goprocam import GoProCamera, constants`
`gpc = GoProCamera.GoPro()`
`gpc.KeepAlive()`
Then restart the stream by typing the following url into a browser (I used Firefox): 10.5.5.9/gp/gpControl/execute?p1=gpStream&c1=restart
Using ffplayer, view the stream with `ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://@10.5.5.9:8554`  [See here](https://github.com/KonradIT/GoProStream/issues/3#issuecomment-342457703)

Thanks for your help!

closed 2019-12-03 by fleather2

tkupek, 2020-03-09:
@fleather2 I have tried both of your solutions but can't get it to work on my Win10 machine.
ffplay never receives any data from the camera.
Do you have any additional tips on how to debug this?

tkupek, 2020-03-09:
Solved it! It was blocked by the Windows Firewall.

il-easteregg, 2020-03-09:
> Solved it! It was blocked by the Windows Firewall.

Can you elaborate? I also run into trouble, reconstructing @fleather2 solution.
Thanks to your advice, tkupek, now I get at least some feedback after doing the gpc.KeepAlive() function. But I still cant access any stream.

saillwhite, 2020-05-28:
Hi @fleather,

I opened this as a top level issue, but since @KonradIT doesn't have a Hero 8, I thought I would ask you. Also thank you very much for these clear and specific streaming instructions!

My problem is that I'm unable to capture the Hero 8's audio stream using ffmpeg. Were you able to somehow magically do so with ffplay? Details below:

### Problem:
The problem is that I'm not able to capture the camera's audio stream with FFmpeg (compiled with libfdk-aac on Linux).

### Details:
I am attempting to direct the stream to a rtmp url. I connect with my rtmp service using their suggested switches:
`
ffmpeg -re -i udp://@10.5.5.9:8554 -c:v libx264 -b:v 1600k -preset ultrafast -c:a libfdk_aac -b:a 128k -s 1920x1080 -x264-params keyint=50 -g 25 -pix_fmt yuv420p -f flv "rtmp://entrypoint"
`

When I do it this way ffmpeg sees four streams, but finds no channels for either audio stream:
`Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709, progressive), 848x480 [SAR 1:1 DAR 53:30], 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc`
`Stream #0:1[0x1100]: Video: h264 ([27][0][0][0] / 0x001B), none(progressive), 24.33 tbr, 90k tbn, 180k tbc`
`Stream #0:2[0x200]: Audio: aac ([15][0][0][0] / 0x000F), 0 channels`
`Stream #0:3[0x201]: Audio: ac3 ([129][0][0][0] / 0x0081), 0 channels, fltp`

It then maps only the first video stream and neither audio stream, thus:
` Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))`

I have tried explicitly setting the audio channels, audio rate, and the sample format of the stream (`-ac 2 -ar 48000 -sample_fmt:0 fltp`). That doesn't help.

I have also tried setting very high analyzeduration and probesize values, hoping that FFmpeg would find channels if given enough time. (`-analyzeduration 15M -probesize 15M`). Still no channels.

The only way I could get FFmpeg to attempt to output an audio channel was via explicit mapping:
`-map 0:v:0? -map 0:a:0?`

By doing this I get the response
`Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))`
`Stream #0:2 -> #0:1 (aac (native) -> aac (libfdk_aac))`

Unfortunately this seems to block the streaming completely, with bitrate= -0.0kbit/s speed=N/A and eventually a Conversion Failed! fatal error.

I realize this is a FFmpeg question and not specific to your work, but I'm curious if you were able to capture an audio stream from the Hero 8, and if so how you did it.

### Setup:
* GoPro Camera(s): Hero8 Black
* Firmware Version: HD8.01.01.60.00
* Steps to reproduce: Put camera in video mode, stop stream via url, start KeepAlive process, start FFmpeg, start stream via url
* Happens every time? : Y