KonradIT/goprowifihack #165: Hero 7/8/MAX Bluetooth commands

Novex, 2019-11-26:
Hi! I’m currently researching the RTMP livestream Bluetooth commands on the 7/8/MAX and was wondering how you found the commands used in [this documentation](https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/Platforms/ArchLinux.md) and your [gopro-ble-py project](https://github.com/KonradIT/gopro-ble-py/blob/master/main.py#L53)? Was it through reversing the GoPro app, or a binary from the GoPro itself, or another method?

I also found the command list at https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces, but they don’t elaborate on how they were obtained and don’t seem to have researched the new BLE commands added after the hero 5.

KonradIT, 2019-11-26:
I found some commands reverse engineering the App yes.

Novex, 2019-11-26:
Thanks for the quick reply - I’ll start there!

If you’ve got any pointers on classes to look at/things to grep for/etc I’m all ears as well