KonradIT/goprowifihack #167: Setting time does not work on Hero5B

datmw, 2019-12-10:
**Problem:**

I'm setting up a script for automation of tasks in our lab and want to ensure that the GoPro has the same time set as the other lab equipment.
Everything else works fine, connection is stable, only setting time results in "Internal Server Error" from the GoPro.

**Details:**

* GoPro Camera(s): Hero5 Black
* Firmware Version: Latest (recently updated)
* Steps to reproduce: Request [](http://10.5.5.9/gp/gpControl/command/setup/date_time?p=%13%0b%10%11%29%2c)

* Happens every time? [Y/N]: Y

Output from browser: 
```
{
"version": "5.00",
"path_info": "command/setup/date_time",
"query_string": "p=%13%0b%10%11%29%2c",
"error_code":-1,
"error_msg": "500 Internal Server Error
",
"function": "gpcontrol_cgi_handler",
"line": 186
}
```

KonradIT, 2019-12-10:
Hmmm I'll investigate on my MAX, thx for letting me know.

datmw, 2019-12-11:
Quick update: Today I tested a different one (we have a total of 3 H5B) which I didn't connect to the app thus didn't install the FW update and it works.
If you want I can have a deeper look into the FW versions tomorrow to help you narrow it down.