KonradIT/goprowifihack #168: Insta360 One X Hack?

TomasHubelbauer, 2019-12-11:
Hi, this is extremely off topic and I'm sorry, but I figured a community for hacking GoPros might overlap with a community for hacking Insta360s, especially seeing as they both have an app which uses wi-fi to connect to the camera. So I'm wondering if you're aware of a similar project to this one, but aimed at the Insta360 One X? I found some stuff online, but it was either outdated or irrelevant. Thank you

KonradIT, 2019-12-11:
Send me one :) 

I've looked at their firmware right after it was released so I could pwn em and have some untested stuff I haven't released.

TomasHubelbauer, 2019-12-11:
Hehe, if you're ever in Prague for a bit, I'll be glad to lend you mine for the duration!

I'll try to capture the wi-fi traffic somehow and play with it to see if I can get something worthwhile to share.

closed 2019-12-11 by TomasHubelbauer

KonradIT, 2019-12-11:
Not in Prague anytime soon but very eager to see what you can come up with.