KonradIT/goprowifihack #170: MAX 360 live streaming

ENQSUWY, 2019-12-27:
**Problem:**

when streaming MAX 360 it's 2 fisheye instead of 360 live streaming...
how to make the in camera stitching streaming 360 live video ? 

**Details:**

connect laptop to MAX wifi
then run ffplay from CMD:
ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://:8554

* GoPro Camera(s): MAX
* Firmware Version:latest
* Steps to reproduce: ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://:8554
* Happens every time? [Y/N]:Y

KonradIT, 2020-01-02:
No idea, you will have to figure that out yourself.

ENQSUWY, 2020-01-02:
thanks, do you know how to read the SD card in MAX on the fly while 360 recording is on? 
any API/command to get the 360 file that I can play back with minimal delay?

KonradIT, 2020-01-10:
>do you know how to read the SD card in MAX on the fly while 360 recording is on?

Not possible.

>any API/command to get the 360 file that I can play back with minimal delay?

Check out the [gopro-py-api](http://github.com/konradit/gopro-py-api) it can do this.

Also, I got MAX live preview working in spherical projection using [QMPlay2](https://github.com/zaps166/QMPlay2) and the gopro-py-api. The sides are not completely stitched and the stream sucks but hey it works.

[image]

[image]