KonradIT/goprowifihack #174: gopro hero 6 Udp issues

fubukai, 2020-01-10:
**Problem:**

After  Url 'http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart' and have {"status":"0" } but i cant URL: udp://10.5.5.9:8554 for streaming 

**Details:**

* GoPro Camera(s):hero6black
* Firmware Version:window 10 
* Steps to reproduce:Live streaming get Url 'http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart' 
* Happens every time? [Y/N]: Y

fubukai, 2020-01-10:
when i try python GoProStream.py 
branch HD4
HERO6 Black
HD6.01.02.01.00
UDP target IP: 10.5.5.9
UDP target port: 8554
message: _GPHD_:0:0:2:0.000000

Recording on camera: False
and i open VLC media player open udp://10.5.5.9:8554 notthing happend

saillwhite, 2020-05-28:
Wondering if you made sure to set your firewall so port 8554 was open to UDP traffic?