KonradIT/goprowifihack #177: Gopro 8 UDP Cam-Commands

Kowal666666, 2020-02-02:
I have some Cam-commands found on google, but it's not complete and some are not working. I can turn on the cammera, like remote do (even when it is turn off), commend for shutter on, works but it is like a button, give it one it will start REC, give the command one again and REC is off. The change mode don't work.

//--------------------- Cam-Commands ----------------------------------------------------------------------
uint8_t PW0[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x50, 0x57, 0x00}; // power off
uint8_t SH1[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x53, 0x48, 0x01}; // shutter 1
uint8_t SH0[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x53, 0x48, 0x00}; // shutter 0
uint8_t CM0[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x43, 0x4D, 0x01}; // change mode <- not working
uint8_t CM1[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x43, 0x4E, 0x01};
uint8_t CM2[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x43, 0x4D, 0x03};
uint8_t wt[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x77, 0x74}; // wait
uint8_t pw[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x70, 0x77}; // hold power on
uint8_t st[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x73, 0x74}; // status?
uint8_t cv[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x63, 0x76}; // cam version
uint8_t OO[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x4F, 0x4F}; // UNKNOWN
uint8_t se[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x73, 0x65}; // UNKNOWN
uint8_t lc[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x6C, 0x63}; // Display image

Do somebody have complete list of command at UDP port, to simulatr the Smart Remote?

* GoPro Camera(s):6,8 Black
* Firmware Version:
* Steps to reproduce:
* Happens every time? [Y/N]:Y

KonradIT, 2020-02-02:
Interesting, how did you get these to work and emulate the smart remote? I tried a while ago to setup a ESP8266 to spawn an AP called HERO-RC-XXXX and send commands, but could not get it to work, only pairing.

Kowal666666, 2020-02-02:
Sorry, i forgot the source. 
https://github.com/sepp89117/GoPro-Multiple-Smart-Remote-ESP8266
I use this as first steps, modified the scetch and going on with some tests. Easiest way would be to turn on camera as AP with UDP pocket (it's working) and switch the arduino to Client and use normal 10.5.5.9 commands as http, but gopro 8 don't turn on the wifi AP, before connecting with gorpo App on smartphone, so http commands/urls has no use for me. I need udp simple commands to operate the gopro, ang project sepp89117 is good direction with gopro 8. I don't know how to scan udp packets from remote, to fully simulate it.

Kowal666666, 2020-02-02:
Tricky thing is, that gopro wifi works like a shit sometimes. Generally i switch the AP on and off, give the connect remote on gopro, and after couple of probes it finally catch the ESP8266 as remote.

Kowal666666, 2020-02-02:
Ok, i found something like this. 
https://github.com/cepoon/GoProRemote
It is what i needed,

Kowal666666, 2020-02-16:
> Interesting, how did you get these to work and emulate the smart remote? I tried a while ago to setup a ESP8266 to spawn an AP called HERO-RC-XXXX and send commands, but could not get it to work, only pairing.

My code, it's working, gopro wakeup in light sleep, but if it is powwered off for longer time it must be powered on by hand. Sorry the code is a little messy, but it's working. I've tried to use hardware serial, but i don't now why it wont receive - it's just sending. Neverless software serial works good. Sometimes ESP8266 is reseting by himself, but i've done it not to interrupt recording if it's on. Values for channel 14 and 8 i'm using to 'communicate' with frysky telemetry passthrough to lua script on X-lite.

#include <ESP8266WiFi.h>
#include <WiFiUDP.h>
#include <SoftwareSerial.h>
#include "mavlink_types.h"
#include "common/mavlink.h"

//--------------------- set MAC for NodeMCU 1.0 (ESP-12E) with 2.5.2 core ---------------------------------
uint8_t ap_mac[] = {0x84, 0xF3, 0xEB, 0x46, 0x0C, 0x1E};// MAC-Adsress of my Smart-Remote
extern "C" void preinit() {
#include "user_interface.h"
  wifi_set_opmode(SOFTAP_MODE);
  wifi_set_macaddr(SOFTAP_IF, ap_mac);
}
//--------------------- heart beat declarations -----------------------------------------------------------
uint8_t stdMsg[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
unsigned int rateI = 1;                                  // durable counter 1
unsigned int rate2 = 1;                                 // durable counter 2
unsigned int rate3 = 0;                                 // durable counter 3
//--------------------- GoPro MAC declarations ------------------------------------------------------------
//uint8_t M1BSSID[] = {0x11, 0x22, 0x33, 0x44, 0x55, 0x66}; //Mac gopro
IPAddress M1IP(0, 0, 0, 0);
//--------------------- Cam-Commands ----------------------------------------------------------------------
uint8_t st[] = {'s', 't'};//status
uint8_t se[] = {'s', 'e'};//session info
uint8_t SH[] = {'S', 'H'};
uint8_t CM[] = {'C', 'M'}; //na stdMsg[13]= 0:'video',1:'photo',2:'burst', 3:'timelapse photo', 7:'settings'
uint8_t cm[] = {'c', 'm'};//camera mode info
uint8_t PW0[] = {'P', 'W', 0x00};//Power off
uint8_t PV[] = {'P', 'V'}; // preview
uint8_t pw[] = {'p', 'w'}; // hold power on
uint8_t wt[] = {'w', 't'}; // wait, send in the network
//--------------------- other declarations ----------------------------------------------------------------
const unsigned int plocalPort = 8383;            // Port der Fernbedienung
const unsigned int cremotePort = 8484;           // Port der Kamera
const unsigned int wifiChannel = 1;             // Channel of my Smart-Remote = 1
const char *ssid_pilot = "HERO-RC-460C1E";//HERO-RC-E423DD";    // SSID of my Smart-Remote "HERO-RC-A1111425435131"
struct station_info *stat_info;
byte packetBuffer[64];                        // buffer to hold incoming and outgoing packets
unsigned long millis_old = 0; 
IPAddress ip(10, 71, 79, 1);                    // IP of my Smart-Remote
IPAddress gateway(10, 71, 79, 1);               // GW of my Smart-Remote
IPAddress subnet(255, 255, 255, 0);             // SM of my Smart-Remote
//---------------------GOPRO declarations ---------------------------
boolean gopro_present = true;
char gopro_mode = 9; //0-video or liveburst, 1-photo or night, 2-burst, 3-timelapse photo
char gopro_rec = 0;
char gopro_time_min = 0;
char gopro_time_sec = 0;
unsigned long millis_gopro = 0;
char gopro_restart = 0;
//----------------------Mavlink-------------------------------------------------
HardwareSerial DebugSerial = Serial;//(14,12); //RX-12, TX-14
SoftwareSerial MavSerial(15,13);

boolean dron_heartbeat = false;
unsigned long millis_command = 0;
unsigned long millis_mavlink_receive = 0;
int dron_status=0, dron_status_old=0;
int dron_ch8=1; //channel to override - status of gopro
int dron_ch14=0, dron_ch14_old=0; //channel to get commands for gopro from telemetry RC - special functions
int dron_alt=0;
int dron_angle=0;

//--------------------- instances -------------------------------------------------------------------------
WiFiUDP Udp;
#define LED 2 //Pin 2 for LED onboard
[... 413 lines cut ...]
//            DebugSerial.print("Channel 15: ");DebugSerial.println(packet.chan15_raw);
//            DebugSerial.print("Channel 16: ");DebugSerial.println(packet.chan16_raw);
//            DebugSerial.println();
//            DebugSerial.print("Chan count: ");DebugSerial.println(packet.chancount);
//            DebugSerial.print("RSSI: ");DebugSerial.println(packet.rssi);
          }
          dron_ch14 = packet.chan14_raw;
          //dron_heartbeat = true;
        }
        break;

      }
    }
  }
}

void Mavlink_CH8_override(int value)
{
  mavlink_message_t msg;
  uint8_t buf[MAVLINK_MAX_PACKET_LEN];

  mavlink_msg_rc_channels_override_pack(255, 25, &msg, 1, 1, 0, 0, 0, 0, 0, 0, 0, value);
  uint16_t len = mavlink_msg_to_send_buffer(buf, &msg);
  MavSerial.write(buf, len);  
}

KonradIT, 2020-03-15:
Hmm, will look into it as I'm dying out of boredom, trapped in this damn box.

MaxwellNewberry, 2020-05-18:
@KonradIT Any update?