KonradIT/goprowifihack #190: Activate Live Streaming Mode on HERO7 / HERO8

Pecacheu, 2020-05-16:
This is different than the 'live stream' mode that's described already for the previous cameras... It's not a preview but a built-in feature of the camera (that you can activate in the app normally) to stream to any URL, you can stream to Twitch, YouTube, etc... Even to OBS using a local RTMP server.

This is the function I'm trying to activate, it's not even mentuined as a mode you can switch to on this page: https://github.com/KonradIT/goprowifihack/blob/master/HERO7/HERO7-Commands.md

However I've figured out the first part at least pretty easily: http://10.5.5.9/gp/gpControl/command/mode?p=2

That puts it into Live Stream mode. But the part I can't figure out is how to tell it what URL to stream to, and activate the stream. I assume that after I give it the URL I'd then use http://10.5.5.9/gp/gpControl/command/wireless/ssid/select?ssid=WIFI_SSID&auth_type=AUTH_TYPE&pw=SSID_PASSWORD to connect it to the WiFi, but that middle step I'm not sure how to figure out...

Is there some way to analyze the network traffic coming out of the app or something?

KonradIT, 2020-05-16:
The new streaming is activated via Bluetooth.

maciek252, 2020-05-20:
Hi KonradIT, could you elaborate more on how to activate the "new streaming"? All the steps?

josebasgmz, 2020-07-15:
Any updates on this topic? Thanks

tubedev2000, 2020-10-12:
> The new streaming is activated via Bluetooth.

any way to caputure/analyze the bluetooth traffic?

DanielnetoDotCom, 2021-09-25:
Looking for a solution