KonradIT/goprowifihack #191: Gopro MAX Power ON command: rather bluetooth and not WOL?

paicolman, 2020-05-28:
I am trying to use the commands on a Gopro MAX, when the cam is on, all works fine, also to turn it off is OK. But it seems that when the camera is OFF, the WiFi gets turned OFF, so a WOL command will not get to the camera.

I am tinkering with an ESP32 to try to turn it back on via Bluetooth, no success so far, but when you install the Gopro App, and connect to the bluetooth of the cam directly (the app does the trust and pairing before, so it´s in the bluetooth list), it turns on promptly.

**Details:**

* GoPro Camera(s): Gopro MAX
* Firmware Version:1.50
* Steps to reproduce: --
* Happens every time? [Y/N]: Y

KonradIT, 2020-05-28:
Yes, MAX powers on when you connect to it via BLE.

paicolman, 2020-05-28:
thanks! Do you have any idea on how to pair to a Gopro? I can connect with the ESP32 when the Gopro is in pairing mode, but I have no idea on what to do to initiate the pairing. Sorry, this is not an "issue" rather a request for help...

KonradIT, 2020-05-28:
Mind sharing your code? I actually got stuck when trying to send BLE commands to the GoPro using the ESP32 board.

paicolman, 2020-05-29:
I'll have to cleanup a bit, at the moment it's just a complete "construction site" with snippets outcommented and stuff (yes, a total mess at the moment). 

But I'll share it sure, although I'm stuck pretty much there, too. I tried with a mac BLE tool to write BLE commands to the camera and it works fine, but not (yet) with the ESP32. I can connect, find the service and characteristics, but still stuck there... I'll post the code here when I've cleaned up a bit.

paicolman, 2020-05-29:
Shoot it got completely reformatted, sry. BTW, this is the log of the script. All seems ok, but writing just won´t work...

`Starting Arduino GOPRO BLE Client application...
BLE Advertised Device found: Name: 846B21C4AAC630A8E9, Address: f3:33:8a:54:d5:de, appearance: 0, manufacturer data: a7050310019944deadbeef000a00ca, serviceUUID: 0000fe07-0000-1000-8000-00805f9b34fb, txPower: 0
BLE Advertised Device found: Name: vívosport, Address: e0:d7:10:34:3e:8e, manufacturer data: 87000a3f, serviceUUID: 6a4e3e10-667b-11e3-949a-0800200c9a66
BLE Advertised Device found: Name: , Address: 71:62:1f:ff:33:a3, manufacturer data: 4c001006131e8f72310c, txPower: 12
BLE Advertised Device found: Name: 846B21C596A630A8E9, Address: f3:33:90:64:56:f4, appearance: 0, manufacturer data: a7050311536f6e6f735f4f584c4559, serviceUUID: 0000fe07-0000-1000-8000-00805f9b34fb, txPower: 0
BLE Advertised Device found: Name: GoPro Cam, Address: db:b0:2f:8f:b1:46, serviceUUID: 0000fe2c-0000-1000-8000-00805f9b34fb, txPower: -22
BLE Advertised Device found: Name: , Address: 45:9b:ee:47:c1:d1, manufacturer data: 4c0010054b1c6c8ba7
BLE Advertised Device found: Name: , Address: 61:ee:fa:3c:79:ec, manufacturer data: 4c0010054b1c6c8ba7
BLE Advertised Device found: Name: , Address: 53:09:19:8b:d4:0e, manufacturer data: 4c001006231aaa2f7d2f, txPower: 8
BLE Advertised Device found: Name: , Address: 76:ba:b9:a7:6a:65, manufacturer data: 4c001006131e18f4c227, txPower: 12
BLE Advertised Device found: Name: , Address: 6b:28:9b:25:89:73, manufacturer data: 4c001006511e344adde4, txPower: 12
************************* WE START CONNECTING HERE *************************
Forming a connection to eb:bb:11:42:32:2f
 - Created client
onConnect has been called
YAY! connected!
 - Found our service
Characteristic is writeable!
*************** CONNECTION OK - 10 sec running  *******************
***************** LETS TRY TO TURN OFF ********************
Trying with a byte array...
Trying with a plain hex number...
Trying with a char code...
************************** WORKED?  *****************************
onDisconnect has been called
`