KonradIT/goprowifihack #192: Hero 8 audio stream not captured by FFmpeg

saillwhite, 2020-05-28:
Hi Konrad - Thank you so much for these amazing tools. I was able to apply them to my new Hero 8 and get it consistently streaming with your KeepAlive function once I opened 8554 for UDP in my firewall. 

### Problem:
The problem is that I'm not able to capture the camera's audio stream with FFmpeg (compiled with libfdk-aac on Linux).

### Details:
I am attempting to direct the stream to a rtmp url. I connect with my rtmp service using their suggested switches:
`
ffmpeg -re -i udp://@10.5.5.9:8554 -c:v libx264 -b:v 1600k -preset ultrafast -c:a libfdk_aac -b:a 128k -s 1920x1080 -x264-params keyint=50 -g 25 -pix_fmt yuv420p -f flv "rtmp://entrypoint"
`

When I do it this way ffmpeg sees four streams, but finds no channels for either audio stream:
`Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709, progressive), 848x480 [SAR 1:1 DAR 53:30], 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc`
`Stream #0:1[0x1100]: Video: h264 ([27][0][0][0] / 0x001B), none(progressive), 24.33 tbr, 90k tbn, 180k tbc`
`Stream #0:2[0x200]: Audio: aac ([15][0][0][0] / 0x000F), 0 channels`
`Stream #0:3[0x201]: Audio: ac3 ([129][0][0][0] / 0x0081), 0 channels, fltp`

It then maps only the first video stream and neither audio stream, thus:
` Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))`

I have tried explicitly setting the audio channels, audio rate, and the sample format of the stream (`-ac 2 -ar 48000 -sample_fmt:0 fltp`). That doesn't help.

I have also tried setting very high analyzeduration and probesize values, hoping that FFmpeg would find channels if given enough time. (`-analyzeduration 15M -probesize 15M`). Still no channels.

The only way I could get FFmpeg to attempt to output an audio channel was via explicit mapping:
`-map 0:v:0? -map 0:a:0?`

By doing this I get the response
`Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))`
`Stream #0:2 -> #0:1 (aac (native) -> aac (libfdk_aac))`

Unfortunately this seems to block the streaming completely, with bitrate= -0.0kbit/s speed=N/A and eventually a Conversion Failed! fatal error.

I realize this is a FFmpeg question and not specific to your work, but I'm curious if you were able to capture an audio stream from the Hero 8, and if so how you did it.

### Setup:
* GoPro Camera(s): Hero8 Black
* Firmware Version: HD8.01.01.60.00
* Steps to reproduce: Put camera in video mode, stop stream via url, start KeepAlive process, start FFmpeg, start stream via url
* Happens every time? : Y

KonradIT, 2020-05-28:
Hi, I don't have a Hero8.

saillwhite, 2020-05-28:
It would almost be worth mailing you one... Two Hero8s are cheaper than an Axis wildlife cam. 

Thanks for your work - I'll post a solution here if (when) I find one!

KonradIT, 2020-05-28:
I do have a MAX, which uses similar software, will see if it has audio. (if you do want to send me a Hero8, go ahead :D)

saillwhite, 2020-05-28:
Thank you for checking that!  I'll start by complaining bitterly on the GoPro forum and repeatedly emphasizing that if they supplied you with a Hero 8 all would be well. Actually my streaming service Dacast wants me to write an article on how to stream with the Hero 8, once I get it figured out. Maybe I'll pressure them as well.

KonradIT, 2020-05-28:
I've checked the MAX and it has audio stream as well. Can listen to it via MPV. Will try to dump audio and video to a file.

saillwhite, 2020-05-28:
My corrupted stream instance did create both MP4 and MOV files, but they were unreadable even by ffmpeg. The videos recorded by the camera have perfect audio and the two streams appear as:
`Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuvj420p(pc, bt709), 1920x1080 [SAR 1:1 DAR 16:9], 45041 kb/s, 59.94 fps, 59.94 tbr, 60k tbn, 119.88 tbc (default)`
and
`Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 189 kb/s (default)`

If that helps at all. I mean if it doesn't work perfectly for you on the first try...

Also, feel free to amuse yourself with my Hero 8 chicken cam, and imagine how much better it will be with sound.

http://www.wickedclever.com/two-rock-chicken-cam/

saillwhite, 2020-05-29:
Thinking the camera is broken. No one else is reporting audio streams with no channels. GoPro support is preposterous. They wanted me to video the issue. I'm going to do a manual update as soon as I get a card reader, and hope that helps. I'll keep you posted...