KonradIT/goprowifihack #196: How to discover the commands

edsonchuck, 2020-06-15:
Firstly, thank you very much for your big effort in making this library.

I would like to ask you, if you know how I could discover the Wifi Commands for a DJI Osmo Action.

I just could discover the IP host of the camera, but I have no idea how to discover the list of commands. I wanna only start recording and stop recording the camera over wifi.

Thank you very much for your attention.

Muchas gracias.

KonradIT, 2020-06-16:
Hi, I've done this using an Android phone running the Packet Capture app, given what the phone sends are HTTP requests. Someone I know told me the commands might be sent via UDP so I don't know if this app will be able to capture them. If not use Wireshark and a rooted phone.

demiantres, 2020-06-16:
I tried to hack the communication without success but I think that the camera uses some Google based UDP protocol (it's been a while an I forgot its name).

KonradIT, 2020-06-16:
Sounds like QUIC?

demiantres, 2020-06-16:
Exactly.

edsonchuck, 2020-06-16:
Thank you very much for your help. At least I know where to start.