KonradIT/goprowifihack #201: Unable to scan wifi networks with HERO7

fpieterse, 2020-08-27:
**Problem:**
I'm trying to connect a HERO7 to a wifi access point using https://github.com/KonradIT/goprowifihack/blob/master/HERO7/HERO7-Commands.md#connecting-to-a-wifi-network

**Details:**

If I do the scan I get {"scan_id":1}
If I then try to list the results it returns a json string with no results:
{
  "scan_id": 1,
  "total": 0,
  "index": 0,
  "index_count": 0,
  "ssid_array": []
}
There are definitely wifi networks that the GoPro should be able to see, I'm not sure if I'm missing a step or a requirement.

* GoPro Camera(s):
HERO7
* Firmware Version:
1.9
* Steps to reproduce:
Camera on, wireless connection on.  Connect computer to camera's access point.
http://10.5.5.9/gp/gpControl/command/wireless/ssid/scan?p=1
http://10.5.5.9/gp/gpControl/command/wireless/ssid/list

* Happens every time? [Y/N]:
yes

mire-dev, 2021-04-08:
Any updates about this issue ?

mire-dev, 2021-04-09:
Hi @fpieterse 

I have managed to get the full list of local wifi networks by sending [http://10.5.5.9/gp/gpControl/command/wireless/ssid/list](http://10.5.5.9/gp/gpControl/command/wireless/ssid/list) multiple times.

Then I can connect to my local SSID (configured IP 10.5.5.X) with _select_ command and see the GoPro in my router network map. Unfortunately, the connection doesn't last long, after 10 seconds, the GoPro is automatically kicked off my local network. 

Has anyone experienced this? 

I have tried with both v7 and v8 editions