KonradIT/goprowifihack #206: Investigate /gpWebcam endpoint

gordonel, 2020-09-23:
Not really a problem, but a suggestion.

GoPro released GoPro Webcam software for HERO8/9 and added an endpoint at `/gp/gpWebcam`. I figured out `/gp/gpWebcam/START` and `/gp/gpWebcam/STOP` commands. `gp/gpWebcam/status` returns an empty array.

Also, it completely ignores any settings set via `gpControl`. I have a HERO9 and can collaborate on this if you'd like

KonradIT, 2020-09-23:
Hi, I have a HERO9 as well, I am preparing a full documentation of gpWebcam and all of the Hero9 commands via BT-LE and WiFi (some WiFi commands have been deprecated).

gordonel, 2020-09-23:
Good to know. Found something in https://github.com/gurgelx/GoProBetaHelper/blob/master/GoProConnector/GoProClient.cs

KonradIT, 2020-09-24:
Working on it... https://twitter.com/konrad_it/status/1309072047272873985

rmoriz, 2020-09-24:
curl http://172.28.128.51/gp/gpWebcam/STOP

curl http://172.28.128.51/gp/gpControl/setting/43/0
curl http://172.28.128.51/gp/gpControl/setting/43/4
curl http://172.28.128.51/gp/gpControl/setting/43/6

curl http://172.28.128.51/gp/gpWebcam/STARTLTP?res=12

gordonel, 2020-09-24:
> curl http://172.28.128.51/gp/gpWebcam/STARTLTP?res=12

This is something new

KonradIT, 2020-09-25:
Don't use LTP, that's for low resolution preview.

KonradIT, 2020-09-25:
Also, I have a PR open in gopro-py-api: https://github.com/KonradIT/gopro-py-api/pull/148

Feel free to fork the project and add your webcam examples there.

So far, my planned usage of gpWebcam is:

- Use FFmpeg and v4l2loopback to create a native /dev/videoX driver, and use NetworkManager dispatch script to do Webcam START when the camera is plugged into my PC, then have an applet to start/stop the FFmpeg thread and turn on/off camera, change resolution, FOV...
- Use FFmpeg to record 30 second video off the GoPro or take a screencap all via USB and upload to my NAS, this will be running on a raspberry pi because I need to monitor my crops

KonradIT, 2020-09-27:
Wow, gpWebcam is truly the gift that keeps on giving. Just made a flask app that displays my media nicely in a web browser and can play back/delete/download. http://github.com/konradit/goprodashboard

This is miles ahead of MTP connection.

KonradIT, 2020-10-11:
Closing this as the information has been added: https://github.com/KonradIT/goprowifihack/commit/4dcd1fb32905d88c8725b380874afe11d822a23a

closed 2020-10-11 by KonradIT

gordonel, 2020-10-11:
Thanks @KonradIT. Do you know if FOV and resolution are the only settings you can change for gpWebcam or is there a way to use manual exposure and, say, flat colour?

KonradIT, 2020-10-11:
Cannot seem to change any setting over Webcam mode when the webcam is active, which is a huge bummer. This is what I get:

````
{
"version": "6.00",
"path_info": "setting/102/6",
"query_string": "",
"error_code":-1,
"error_msg": "500 Internal Server Error
",
"function": "gpcontrol_cgi_handler",
"line": 190
}
````

Perhaps GoPro can add custom ProTune controls to Webcam because I want to lower the max/min ISO to 100-400 like I always have on video mode, have a warmer white balance and use flat color.

gordonel, 2020-11-07:
it seems to be the same on the latest firmware. Can you verify?