KonradIT/goprowifihack #208: Reconnecting Hero 8 without starting pairing mode

demiantres, 2020-09-26:
I use the commands from here: https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/Platforms/RaspberryPi.md

The following works:
1. On the Hero 8: Connections -> Add GoPro app
2. sudo bluetoothctl
3. scan on
4. trust DE:71:28:EE:A7:FD
5. pair DE:71:28:EE:A7:FD
6. sudo gatttool -t random -b DE:71:28:EE:A7:FD -I
7. connect
8. char-write-req 2f 03160101

But when I turn off the GoPro or exit the pairing mode then I cannot reconnect. Instead I have to go back to the pairing mode each time on the Hero 8. Is there some command missing?

KonradIT, 2020-09-28:
Once you do step 5, disconnect, exit bluetoothctl, exit pairing mode on H8, turn off Wireless connections, turn back on, proceed to step 6.

demiantres, 2020-09-29:
Thanks. Maybe it's a problem with my Bluetooth dongle (Asus BT400) because it does not work.

Leeleman, 2020-10-26:
Same problem with Hero 7 Black and Jetson Xavier NX. 
I can connect to GoPro only then it in pairing mode.

Then pairing mode off (bluetoothctl):
[bluetooth]# connect D6:C4:1A:2C:D1:FD
Attempting to connect to D6:C4:1A:2C:D1:FD
[CHG] Device D6:C4:1A:2C:D1:FD Connected: yes
[CHG] Device D6:C4:1A:2C:D1:FD Connected: no
Failed to connect: org.bluez.Error.Failed

Then pairing mode on (bluetoothctl):
[bluetooth]# connect D6:C4:1A:2C:D1:FD
Attempting to connect to D6:C4:1A:2C:D1:FD
[CHG] Device D6:C4:1A:2C:D1:FD Connected: yes
Connection successful

Then pairing mode off (gatttool):
someuser@somehost:~$ sudo gatttool -t random -b D6:C4:1A:2C:D1:FD -I
[D6:C4:1A:2C:D1:FD][LE]> connect
Attempting to connect to D6:C4:1A:2C:D1:FD
Error: connect error: Function not implemented (38)

Then pairing mode on (gatttool):
someuser@somehost:~$ sudo gatttool -t random -b D6:C4:1A:2C:D1:FD -I
[D6:C4:1A:2C:D1:FD][LE]> connect
Attempting to connect to D6:C4:1A:2C:D1:FD
Connection successful

Some information from systemctl status bluetooth:
bluetoothd[5707]: src/adapter.c:connected_callback() hci0 device D6:C4:1A:2C:D1:FD connected eir_len 23
bluetoothd[5707]: src/adapter.c:dev_disconnected() Device D6:C4:1A:2C:D1:FD disconnected, reason 0
bluetoothd[5707]: src/adapter.c:adapter_remove_connection()
bluetoothd[5707]: plugins/policy.c:disconnect_cb() reason 0
bluetoothd[5707]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr D6:C4:1A:2C:D1:FD type 2 status 0xe
bluetoothd[5707]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[5707]: src/device.c:device_bonding_failed() status 14
bluetoothd[5707]: src/adapter.c:resume_discovery()

Could this issue be controller related?

Leeleman, 2020-10-29:
I don't know how useful this information will be, but I found that when GoPro Hero 7 Black is in pairing mode, I can connect to it from device without trust or pairing. At the same time, connections to the camera when the pairing mode is off have the same result, regardless of whether the camera was previously paired with the device or not.
I came to the conclusion that for some reason the pairing is not performed on the camera side, because the keys on the device side are recorded.

Leeleman, 2020-11-03:
I have successfully connected Jetson Xavier NX to GoPro Hero 7 Black using USB bluetooth adapter BURO BU-BT40A. Wake up via bluetooth working too.

mire-dev, 2021-04-13:
Hi there, 

I'm facing the same issue with both GoPro Hero 7 and 8. I'm trying to bond these devices with Ubuntu and Windows without success.  Otherwise, I can easily wake up GoPros with nRF Android app on my mobile phone. 

Should I have to consider that this is a Bluetooth interface problem from my laptop? Do you recommend to try with different external BT adapters?

marcotidei, 2024-02-18:
> Once you do step 5, disconnect, exit bluetoothctl, exit pairing mode on H8, turn off Wireless connections, turn back on, proceed to step 6.

After a few years, someone is still struggling with this problem. I follow what suggested but this means that the GoPro never fully pairs with my RPi Zero, so while it is in paring mode I can send commands. Still, without completing the pairing I cannot for example wake the camera because the Bluetooth LE remains off. Any suggestions?