KonradIT/goprowifihack #212: Live feeding to PC(internet connected) from GO Pro

tskarthikeyann, 2020-11-12:
First of all, Thank you very much for all your wonderful work. 

**What I am trying to do:**
I want to get the Livestream out of **Gopro 8**. 
Right now, I am doing it by setting Gopro for RTMP and on the pc side, using the Mona server. 

**What’s the challenge now?**
While previewing the live stream, my pc must be connected with the internet as well. 
Also, I am not allowed to use the GoPro mobile App. 
I believe, GoPro acts only as an Access Point for PCs to connect through Wifi. 

**What am I looking for?** 
1. How to Livestream(HTTP or UDP or whatever)  from gopro8 to a pc which must be connected with the Internet?  
Or 
2. Using Bluetooth commands, is there a way to automate what you see in the screenshot without GoPro or any mobile app? ( Connecting Go Pro with a specific access point, setting GoPro with RTMP URL, and start live streaming ) 
https://drive.google.com/file/d/1S3UI7eFL4pAejRt7MNnAFv6t68Mb-qvp/view?usp=sharing ( Screenshot)

demiantres, 2020-11-12:
Might this work for you: https://www.toolsforgopro.com/live ?

tskarthikeyann, 2020-11-12:
Wow.. https://www.youtube.com/watch?v=EvGU1MXpZ1E&feature=youtu.be - Yes, I think that's the perfect solution. Thank you. 
As i want to integrate this rtmp contoller as part a part of my existing application, is there a way to do it my own ? References, if any please?