KonradIT/goprowifihack #219: SmartRemote 10.71.79.1:8484 - use for file transfer?

habakuk40, 2021-01-12:
High KonradIT!

I am happy to learn that your amazing site is as up as it was 2 Years ago.
You won't remeber me but I was trying to set up a gopro as dashcam.
We went pretty far by now the last 2 Years, HW wise:
[image]

On our way of building 6 differnt pcbs, I figured, that you were dam right - we stumbled over a ton of issues - last but most important: stablized, central Powerbank based, car battery independent PowerSupply (get up to 6 Amps power load independent stabilized by stepupconverters from 5,1 - 6V (see upper left section of picture)) for the central Controll-unit of this FleetMaster, an Odroid in this case (Ubuntu). (you can use raspi too).

However, we would be more than happy to share our Eagle board filez with you, just in case....

We added a GPS support, Gsensors and an complete homemade IO Interface connected to the GPIOs of Odroid/Raspi to read/write digital and analoge Sensors/Signals .

However, thanks to your research, we are going to controll the gopro by Odroids Ubuntu over http-nodes of NodeRed.

However, my question: I stumbled over:
https://goprohero.readthedocs.io/en/latest/Wifi%20Research/
https://github.com/theContentMint/GoProRemote
https://github.com/cepoon/GoProRemote 

an I got my PC and the Hero5 all connected under the 10.71.79.0/24 subnet with the Hero5 as 10.71.79.1 as recommended.
So I am trying to build up a stable conenction in the "SmartRemote-World" (for file transfer), independend and parallely to the 10.5.5.9 world (for controlling....)

The clients can see each other but I cannot ping , nor netcat to 10.71.79.1 on 8484 or ANY other Port.
We checked it a 1000times - its not firewall issue of the router.

What are we doing wrong?
How to realize an unattened file transfer

greetz
HABAKUK
#never dump a prohject
**Details:**

* GoPro Camera(s): GoPro5black (2018)
* Firmware Version:2.60
* Steps to reproduce:
* Happens every time? [Y/N]:y

KonradIT, 2021-01-13:
Hi, I don't know if the smart remote connection can even do file transfer (you mean getting media from /videos/DCIM?). Pinging @sepp89117 who is working on smart remote code and might now this.

sepp89117, 2021-01-13:
Hi, in remote control mode (RC) the cameras connect to the access point (AP) of your RC. The command transmission runs over UDP. In "HTTP mode" (I'll call it that) you connect to the AP of a camera. The command is of course transmitted via HTTP.
You can control the cameras and receive information from them via HTTP and UDP. However, file transfers only run via the "HTTP mode".
In my project (GoEasyPro and
GoPro-Multiple-Smart-Remote-ESP8266) I control several cameras at the same time in RC mode and can individually watch the videos on the SD cards in HTTP mode.

Hope that helps.

habakuk40, 2021-01-13:
Thank you for your input @KonradIT  and @sepp89117 !!

Just to be absolutely sure:
A) the 10.71.79.1 line is UDP only and therefore only suitable for controlling the HERO (what the original SmartRemote does in fact), in that case the camera is client.

B) in the 10.5.5.9 line its UDP and HTTP, which is suitable for everything , filetransfer included (konradit......) the camera acts as AP and connects to a client (cell phone app), as long as the BLE background pairing works out (gattool....) 
So my question is:
On the one hand I MUST controll the camera over an odroid (ubuntu, nodered) with direkt wifi/BLE connection to the camera and simultaneously grep (download) the moviefiles from /videos/DCIM by and to another client (windows client behind Linksys1900ACSV2).

1.) Is this possible to realize the whole connetivitiy over the 10.5.5.9 line?
My fear, but dont know yet: no, because camera only accepts on client as AP in 10.5.5.9 mode who must be BLE accredited, and
additionaly even if the camera accepts two cliennts at the same time with two paired BLE connections, I cannot provide this with the windows client behind the router. :-(
Please tell me that I am wrong.
@KonradIT : Sorry its about 2 years ago that I was deeply in your core compentence topic here: is it possible for a client to have full http-functionality over 10.5.5.9 without having his active BLE pairing to the camera?
Seems I remeber thet the (instable) BLE connection was just working as an add-on to the http 10.5.5.9 connection on top to be able to wake up the camera when wife shut downor something like that ... if I remeber right.

2.)  @sepp89117 What if I set it up that way: 
on the odroid:
- realize the "controlling the HERO" as a AP 10.71.79.0 over UDP to 10.71.79.1 (camera) (I never got that working so far: cannot netcat 10.71.79.1:8484, dont know what I am doing wrong.)
- setup BLE paring (gattool) for 10.5.5.9 wifi reovery if possible (nodered)
-
and on the windows client: 
- set up a client 10.5.5.x to be able to http top the camera.

Does this sound like a plan to you guys?
@KonradIT @sepp89117

sepp89117, 2021-01-14:
Hi, there are enough commands known for the RC mode. The best thing to do is take a look at my project. Aside from the YY commands, I've got everyone I know in there. I just commented out the ones I don't use.
https://github.com/sepp89117/GoPro-Multiple-Smart-Remote-ESP8266/blob/master/GoProRC_UDP2_HT.ino
If you miss something about commands, ask me about them.

It is just very important how you set up your AP! You have to change the MAC address, the IP (10.71.79.1 is RCs IP, Cams are .100+), the SSID and the network has to be open (without encryption). Once the cameras are connected, you need to keep the connection alive through regular communication. If a parameter is incorrect, it will not work.

habakuk40, 2021-01-19:
hi Sepp!
I shipwreck already at translating your AP byte-array-mac ({0x84, 0xF3, 0xEB, 0xE4, 0x23, 0xDD} to  normal hex mac.

According to  my sources sofar the mac of the smartremote-AP has to be something like
D8:96:85:xx:yy:zz

where as xx:yy:zz have to be the partz og the AP ssid HERO-RC-xxyyzz!

right?

sepp89117, 2021-01-20:
Hi
The data in my code is taken from a real smart remote and i can't find any connection between the SSID and the MAC.  However, I have already had HERO-RC-800001 with the MAC ending 00:00:01 running.  I discovered and tested HERO-RC-800000 in firmware data of a GoPro.  This also works, but the camera forgets this pairing after a reboot.

PS: What is the difference between a byte array MAC and a hex MAC, besides the format?

habakuk40, 2021-01-20:
ok going for it.

sepp89117, 2021-01-26:
Hi @habakuk40 
Take a look at https://github.com/sepp89117/GoPro_ESP32_TFT_Touch-Control/
Maybe that will help you.

habakuk40, 2021-02-17:
@sepp89117 

hi there!

Finally I do habe my working hostapd /dnsmasq setup on my odroid C2!!

And I have my MAC address under controll under Ubuntu 18.04 with WifiModule5A under rtl8812au thanks to
sudo modprobe 8812au rtw_initmac=new mac (!) which was a hell of a ride!

so my AP holds:
IP4: 10.71.79.1
DNS: 255.255.255.0
MAC: 86:F3:EB:E4:23:DD

my DNS server sets my GoPro on a lease with 10.71.79.100

My Go pro connects but although I changer the AP's MAC to the address you recommended (see above) it forgets the pairing after reboot the Gopro and even cannot connect at all anymore.

I have to reset all connections on the gopro to make it connect again ...

What am I doing wrong - please help!

sepp89117, 2021-02-18:
Hi @habakuk40 
did you try MAC d8 96 85, 80, 00, 01 together with SSID HERO-RC-800001 ? The SSID (name of your wifi) must match the MAC address.

habakuk40, 2021-02-18:
Hi @sepp89117 

"did you try MAC d8 96 85, 80, 00, 01 together with SSID HERO-RC-800001 ? The SSID (name of your wifi) must match the MAC address."

this is somehow contradictove to your first post:
The data in my code is taken from a real smart remote and i can't find any connection between the SSID and the MAC.  However, I have already had HERO-RC-800001 with the MAC ending 00:00:01 running.  I discovered and tested HERO-RC-800000 in firmware data of a GoPro.  This also works, but the camera forgets this pairing after a reboot.

However:
neither: your code-suggestion
86:F3:EB:E4:23:DD -->HERO-RC-A1111425435131

nor:
d8 96 85, 80, 00, 01 -->HERO-RC-800001

work!

initial pairing : success!   but
issues:
1.)forget after reboot GoPro
2.)no manual reconnect without reset all connections!

what should I do?

sepp89117, 2021-02-18:
Sorry, I misunderstood myself.  I cannot see any connection with 86: F3: EB: E4: 23: DD -> HERO-RC-A1111425435131.  Nevertheless it has to belong together.  With 01 at the end it works too.  Your fault must be elsewhere.  If you look in the camera after the reboot, is there still smart remote listed under connections? I assume: yes.  Then the camera no longer wants to connect because it does not recognize the remote as such.  Check whether your AP assigns an IP when you turn on the camera.if so, something is wrong with the program.  Are you sending a heartbeat  You have to send various commands rhythmically.

habakuk40, 2021-02-18:
@sepp89117 

... working on the dhcp of the AP.

When the gopro goes offline, the remaining unused lease in the arp-chache seems to be the problem - working on it....
if I clear the arp cache I can reconnect without resetting the gopro

@sepp89117  please tell me about the "various commands rhythmiucally..." so that I can use them to prevent going the gopro offline while it is still switched on...

sepp89117, 2021-02-18:
@habakuk40 
As soon as the camera has received an IP, you have to send it a message via UDP about every 1.5 seconds.  The camera replies with a repetition of the message and appends a 1 for "error" or a 0 for "okay".
You can also increase the rhythm, up to a limit of approx. 600ms, if you need information faster.
You can find all possible messages in my code.  You can also find what I send as a heartbeat there.

habakuk40, 2021-02-18:
@sepp89117 

hey!

our achievement of the day is  to set shutter on/off over Ubuntu 18.04 with NodeRed's UDP nodes.

- Gopro doesnt show paired with smart remote "Not connected" after reboot but is IN fact reacting propperly to the commands. I am surprised.
- reáction is somewhat quicker and exact , when compared to the http interface.
- would have been helpfull to mention that I have to implement Counter to byte 9 and 10 into my UDP requests in order to get any reaction of the cam ;-)
- Sofar I cannot confirm the need of a heartbeat yet  - camera stays tuned in - don't know yet how long - i am not enough into this I think, however.

ISSUE sepp, please:

If I compare the amount of the commands of  (konradIT-http)

https://github.com/KonradIT/goprowifihack/blob/master/HERO5/HERO5-Commands.md       (konradIT-http)

with your (seppUDP)

https://github.com/sepp89117/GoPro-Multiple-Smart-Remote-ESP8266/blob/master/GoProRC_UDP2_HT.ino (seppUDP)

I miss a tremendous amount of commands!

I not really getting further with 
//and a lot of YY commands, take a look at https://orangkucing.github.io/Hero4_I2C_Commands.html

Can you provide a detailed UDP Command list which is comparably detailed to the KonradIT http list as you mentioned earlier 
 in this thread?
I mean there is not even a TAG - udp- Command.
This would be really important for us!!!!

greets adn many thanks in advance

Peter

P.S.: R U German? (~Sepp | Josef) ;-)

sepp89117, 2021-02-19:
@habakuk40 
Hi, yes i am german.
I'm not exactly sure what you mean by TAG command.  What should the cmd do?

If I find the time to list the commands, I'll do it. However, this can take time.  The available commands via UDP are significantly fewer than via http.  The YY commands expand this significantly.  Unfortunately there is no complete list either.  I found what I need with a lot of effort using the @orangkucing foundation and documentet it in my code.

habakuk40, 2021-02-19:
@sepp89117 
servus!

for example - in the https world:
Tag moment:
Sends tag command: http://10.5.5.9/gp/gpControl/command/storage/tag_moment

it "stars" or flags (marks as favourite) a video-file  while itis taken.

that is really important.

I completely understand that it takes a lot of time to list the UDP commands, but as they are a really important necessarity for me I am dependent on this.

Is there any help I can do for you?

 https://orangkucing.github.io/Hero4_I2C_Commands.html

is more or less "bahnhof" ;-) to me.

In case I can/should help you to expand the UDP command list please let me know or tell me HOW to do it.

sepp89117, 2021-02-20:
Maybe you can contact me via gmail.  I have the same username there as here.

sepp89117, 2021-02-21:
@habakuk40 
Hi, I have summarized almost all UDP commands that I have and published them here:
https://github.com/sepp89117/GoPro_Hero-5_UDP_commands
There is also an Excel file.