KonradIT/goprowifihack #223: Question about stream quality

vinnn-90, 2021-01-21:
**Problem:**

Hey folks,

I finally got the video stream with ffplay working with my GoPro Hero 2018.
I am still getting lots of error reports but the video is kinda stable.

It just looks totally horrible, like 144p quality.

I tried upping the Stream Bitrate but then ffplay wont open the stream and i get even more errors.

Is there anything I can do about the quality?

**Details:**

- GoPro Hero 2018
- ffmpeg version 2021-01-12-git-ca21cb1e36-full_build-www.gyan.dev Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10.2.0 (Rev6, Built by MSYS2 project)
- ffplay -fflags nobuffer -f:v mpegts -probesize 16192 udp://10.5.5.9:8554
- Cam is connected via Wifi PCI Card 
- Windows 10

**Any help is much appreciated ! <3**

Here is the cmd log when I use 1 Mbps:
`Microsoft Windows [Version 10.0.19042.746]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\Vinh>ffplay -fflags nobuffer -f:v mpegts -probesize 16192 udp://10.5.5.9:8554
ffplay version 2021-01-12-git-ca21cb1e36-full_build-www.gyan.dev Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 10.2.0 (Rev6, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-lzma --enable-libsnappy --enable-zlib --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-libbluray --enable-libcaca --enable-sdl2 --enable-libdav1d --enable-libzvbi --enable-librav1e --enable-libsvtav1 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libaom --enable-libopenjpeg --enable-libvpx --enable-libass --enable-frei0r --enable-libfreetype --enable-libfribidi --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-libglslang --enable-vulkan --enable-opencl --enable-libcdio --enable-libgme --enable-libmodplug --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libshine --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libilbc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-ladspa --enable-libbs2b --enable-libflite --enable-libmysofa --enable-librubberband --enable-libsoxr --enable-chromaprint
  libavutil      56. 63.100 / 56. 63.100
  libavcodec     58.116.100 / 58.116.100
  libavformat    58. 65.101 / 58. 65.101
  libavdevice    58. 11.103 / 58. 11.103
  libavfilter     7. 95.100 /  7. 95.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100
[h264 @ 000001b0f25219c0] non-existing PPS 0 referenced  0B f=0/0
    Last message repeated 1 times
[h264 @ 000001b0f25219c0] decode_slice_header error
[h264 @ 000001b0f25219c0] no frame!
[h264 @ 000001b0f25219c0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 000001b0f25219c0] decode_slice_header error
[h264 @ 000001b0f25219c0] no frame!
[h264 @ 000001b0f25219c0] non-existing PPS 0 referenced  0B f=0/0
    Last message repeated 1 times
[h264 @ 000001b0f25219c0] decode_slice_header error
[h264 @ 000001b0f25219c0] no frame!
[mpegts @ 000001b0f250fd40] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
Consider increasing the value for the 'analyzeduration' (0) and 'probesize' (16192) options
Input #0, mpegts, from 'udp://10.5.5.9:8554':
  Duration: N/A, start: 289.255633, bitrate: N/A
  Program 1
    Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709, progressive), 432x480 [SAR 2:1 DAR 9:5], 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc
    Stream #0:1[0x1100]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 120 kb/s
    Stream #0:2[0x200]: Unknown: none ([128][0][0][0] / 0x0080)
[swscaler @ 000001b0f9b920c0] deprecated pixel format used, make sure you did set range correctly
[mpegts @ 000001b0f250fd40] Packet corrupt (stream = 0, dts = 26279253).
[mpegts @ 000001b0f250fd40] Packet corrupt (stream = 0, dts = 26885859).
[mpegts @ 000001b0f250fd40] Packet corrupt (stream = 0, dts = 27459432).
[h264 @ 000001b0f9ad6dc0] left block unavailable for requested intra mode
[h264 @ 000001b0f9ad6dc0] error while decoding MB 0 15, bytestream 11552
[h264 @ 000001b0f9ad6dc0] concealing 454 DC, 454 AC, 454 MV errors in I frame
[mpegts @ 000001b0f250fd40] Packet corrupt (stream = 0, dts = 28036008).
[h264 @ 000001b0f9bfae80] left block unavailable for requested intra4x4 mode -1
[h264 @ 000001b0f9bfae80] error while decoding MB 0 24, bytestream 4076
[... 135 lines cut ...]
[h264 @ 0000025aff0fbb40] no frame!
[h264 @ 0000025aff0fbb40] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000025aff0fbb40] decode_slice_header error
[h264 @ 0000025aff0fbb40] no frame!
[h264 @ 0000025aff0fbb40] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000025aff0fbb40] decode_slice_header error
[h264 @ 0000025aff0fbb40] no frame!
Input #0, mpegts, from 'udp://10.5.5.9:8554': 0KB sq=    0B f=0/0
  Duration: N/A, start: 7.530667, bitrate: N/A
    Stream #0:0[0x1100]: Audio: aac (LC), 48000 Hz, stereo, fltp, 128 kb/s
    Stream #0:1[0x1011]: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 432x480 [SAR 2:1 DAR 9:5], 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc
[swscaler @ 0000025a88534680] deprecated pixel format used, make sure you did set range correctly
[mpegts @ 0000025aff08fd80] Packet corrupt (stream = 1, dts = 1087086).
[mpegts @ 0000025aff08fd80] Packet corrupt (stream = 1, dts = 1657656).
[h264 @ 0000025a88506840] concealing 331 DC, 331 AC, 331 MV errors in I frame
[mpegts @ 0000025aff08fd80] Packet corrupt (stream = 1, dts = 1678677).
[mpegts @ 0000025aff08fd80] Packet corrupt (stream = 1, dts = 2210208).
[h264 @ 0000025a88507000] concealing 563 DC, 563 AC, 563 MV errors in I frame
[mpegts @ 0000025aff08fd80] Packet corrupt (stream = 1, dts = 2762760).
[h264 @ 0000025a88506840] cabac decode of qscale diff failed at 8 25
[h264 @ 0000025a88506840] error while decoding MB 8 25, bytestream 5626
[h264 @ 0000025a88506840] concealing 176 DC, 176 AC, 176 MV errors in I frame
  31.71 A-V:  0.110 fd= 247 aq=    0KB vq=    0KB sq=    0B f=0/0`

simspace-dev, 2021-04-11:
@vinnn-90, did you ever figure out this problem. I'm dealing with "Packet corrupt" issues too.

vinnn-90, 2021-04-11:
If I remember correct, the errors do not matter. The quality is restricted when u try to pull the stream from the cam anyway. I have stopped trying because it seems there is no way to pull the full quality video stream :(