KonradIT/goprowifihack #229: HERO9 - bluetooth commands gatttool not working

mixmaxme, 2021-03-12:
**Problem:**

I trusted + paired + connected my GoPro Hero9 via bluetoothctl, then disconnected, exit.
Opening 
sudo gatttool -t random -b <bl-mac-address> -I
connect (successfull)
char-write-req 2f 03170101
--> Error: Characteristic Write Request failed: Attribute can't be written

**Details:**
The GoPro was never registered anywhere else (and is reset) / also after another facotry reset

* GoPro Camera(s): Hero9 black
* Firmware Version: 01.00
* Steps to reproduce: see above
* Happens every time? [Y/N]: Y

Thanks in advance!

KonradIT, 2021-03-12:
Hi, char changed. https://github.com/KonradIT/goprowifihack/blob/master/HERO9/Bluetooth-Commands.md

jacobstasiewicz, 2021-07-30:
hello @mixmaxme did this work for you even after char was changed? I am trying to connect my gopro via gatttool, and am having some trouble. I am a newbie to gatttool but was able to connect it using bluetoothctl