KonradIT/goprowifihack #239: BLE RECONNECT not working  - missing 4 Key values in /var/lib/bluetooth/${HCI_MAC}/${GOPRO_MAC}/info

habakuk1979, 2021-04-24:
@KonradIT 

hey back again!

I do have the exact problem you mentioned here:
---------------------------
Make sure the negotiated pairing keys are stored in /var/lib/bluetooth/${HCI_MAC}/${GOPRO_MAC}/info - there should be 4 sections titled: RemoteSignatureKey, LocalSignatureKey, LongTermKey, SlaveLongTermKey. Without these, you technically didn't complete pairing and you would not be able to access the GoPro over BLE until you reset all connections on the GoPro to start the pairing process again.
---------------------------

Where do I find the negotiated paring keys?

habakuk1979, 2021-04-24:
ahh, git it!

one must Enable "LE-mode" of the "bluetooth.service".

repair

et voilá:

the keys are in the /var/lib/bluetooth/${HCI_MAC}/${GOPRO_MAC}/info!

habakuk1979, 2021-04-24:
@KonradIT 

Bit still I cannot reconnect.

back to my old bug: bluetoothctl re-connects for like 1 second but the disconnects again.

cannot achieve stable BLE connection although there are all four keys in the info file AND ble (--experimental) enabled.

what am I doing wrong?

habakuk1979, 2021-04-30:
@KonradIT 
Which platform in Linux (Flavour) and which kernel were you using doing your bluetooth connection attemts?
(uname -a)

I am fairly old on: Ubuntu 18.04 kernel 3.16.85+

this has several reasins (wifi connectivity, cmake of other drivers, ...)

However, I'd appreciate your assinstance in this case please, if possible.

KonradIT, 2021-04-30:
I've been using Arch Linux on a recent kernel and Raspbian OS with whatever kernel it ships with.

habakuk1979, 2021-04-30:
@KonradIT ok.
do you have any other idea in terms of configuring BLE ("main.conf" or "/lib/systemd/system/bluetooth.service") which can knock out my connection - stability problem?