KonradIT/goprowifihack #240: What's the most comparable gopro for use with WIFI, API or BLE API? Focus: Triggering and downloading time-lapse

pgregg88, 2021-05-04:
**Problem:**

I'm experiencing different challenges using various editions of @KonradIT 's great work with my HERO3 (too old) and a HERO9 (too new?).

I'll likely be returning my HERO9 and going with a different model, but I'm open to all recommendations.
Is there a specific HERO that folks have the most success with?

Here are my characteristics.
- Main function: Kickoff a daily sunrise, mid day, sunset and night-time time-lapse (unique settings required)
- single camera
- must be able to download/transfer images to a linux server via WIFI or USB (then delete off SD card)
- always powered via usb cable
- access is possible by WIFI, BLE and/or USB (not a requirement to use all three, but is possible)
- connected to a house, so access is inconvenient, but possible.

Thanks in advance.

KonradIT, 2021-05-04:
Hi, you're going to run into trouble sooner or later with most GoPros if you plan to have the camera always shooting. For a issue-proof setup, I'd use the Hero7/8/9 with GoPro Labs firmware, with [power on via USB power QR code](https://gopro.github.io/labs/control/usb/), and the RPI could use [this program](https://github.com/mvp/uhubctl) to turn off/on the USB hub. Then, for each time of day, the Pi turns on the USB hub, then I would use Bluetooth for settings change + shutter, with fallback to WiFi. USB control is limited, and with the Hero9 USB control you cannot change the settings. 

For downloading, depending on what camera, WiFi or USB.

as for deleting files, use WiFi.

pgregg88, 2021-05-06:
Thanks. Great advice.