KonradIT/goprowifihack #252: How to capture commands

TheGreyDiamond, 2021-07-27:
**Problem:**

I want/need to capture new commands from the GoPro QUIK app. I wanted to use Packet Capture, however, the moment I enable any kind of a VPN (like Packet Capture or Netguard) GoPro QUIK losses connection to the camera. Any idea why this happens?

* GoPro Camera(s): GoPro Hero 7 black

KonradIT, 2021-07-27:
Start the VPN after you connect to the camera on the app.

closed 2021-07-27 by KonradIT

TheGreyDiamond, 2021-07-27:
Last time I tried it also failed. I had the GoPro app and Packet capture open next to each other on my tablet and the moment I start the packet capture the GoPro app shows a dialogue saying it has lost connection to my GoPro. Even tho I was seeing the live preview (until I started the packet capture). So I was connected.

TheGreyDiamond, 2021-07-31:
So, any idea?