KonradIT/goprowifihack #254: Starting and stopping video with BLE requests (HERO9)

jacobstasiewicz, 2021-08-10:
**Problem:**

I am trying to find the bluetooth requests to start and stop the recording of my gopro 9, I already have figured out how to control video settings with gatttool. Is there a cheat sheet anywhere? Or could someone explain to me how to find the requests for certain commands, I am new to this and all I can figure out how to do is change the camera resolution.  This page https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/ArchLinux.md is not working 

**Details:**

* GoPro Camera(s): Gopro 9
* Firmware Version: 01.00
* Steps to reproduce:  NA
* Happens every time? [Y/N]: Y

jacobstasiewicz, 2021-08-10:
I having a tough time figuring out how to convert the Requests section from https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/bluetooth-api.md to bluetooth requests

KonradIT, 2021-08-10:
To start and stop video send the following byte array:

Start: `3,1,1,1`

Stop: `3,1,1,0`

And to use with gatttool, copy the command as specified here: https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/Platforms/ArchLinux.md

jacobstasiewicz, 2021-08-12:
Thanks I was having trouble finding this page

jacobstasiewicz, 2021-08-12:
I am still having a tough time I can connect to my camera and change settings with ```char-write-req __________ ``` perfectly fine but when I try to use ```char-write-req 38 03010101``` to start recording, gatttool reads " characteristic value was written successfully "but then the camera wont record

KonradIT, 2021-08-12:
https://github.com/KonradIT/goprowifihack/blob/master/HERO9/Bluetooth-Commands.md