KonradIT/goprowifihack #256: I need to know the status of the GoPro (know if it is recording or not, and know if the camera is off or not)

wilbercl, 2021-08-31:
**Problem:**

I need to know the status of the GoPro (know if it is recording or not, and know if the camera is off or not)

**Details:**
Hello, I am developing a script that monitors the GoPro cameras, I need to know each status of the cameras, for example, if it is being recorded, if the camera is off or on. For this, I think I need to know the UUID of each state.

For example to know the battery percentage I am using the following UUID: 00002a19-0000-1000-8000-00805f9b34fb.
So I do something like this:
value = bytes (await client.read_gatt_char (BATERRY_LEVEL))
valueinutf = str (ord (value)) + "%"
print ('Battery Level:', valueinutf)

Work with GoPro hero 8 Black

Thanks!!

KonradIT, 2021-08-31:
Check out https://gopro.github.io/OpenGoPro/ble#status-codes

wilbercl, 2021-09-01:
Hola @KonradIT
Thank you very much for answering, I have been looking at the page you sent me, but they do not say which are UUID of the states that I want to know (know if the GoPro are recording and know if they are off or on). In the link, I see that they explain what each label means, how do I get those labels? Will these keys always appear as output with some value? If you could help me I would appreciate it!
Muchas Gracias!

wilbercl, 2021-09-01:
Could you do something similar to what was stated at the beginning of my question, to know if the camera is recording or not, and to know if it is on or off? knowing the UUID of each command, for example:
await client.read_gatt_char (00002a19-0000-1000-8000-00805f9b34fb)

KonradIT, 2021-09-01:
The documentation I posted earlier already explains how to get the battery status, it's not shown on a characteristic, rather a notification.

closed 2021-09-01 by KonradIT

wilbercl, 2021-09-01:
ok, thanks, I will try to do it as they say.

Fougl, 2024-03-24:
> The documentation I posted earlier already explains how to get the camera status, it's not shown on a characteristic, rather a notification.
> 
> _Sent from my HD1900 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Hi Konrad, do you know how to get battery level on Max?