KonradIT/goprowifihack #259: Naming files video setting gopro

wilbercl, 2021-09-13:
**Problem:**
I would like to know if there is any way to change the name of the files created by the GoPro hero 8 when taking a video. By default, the Gopro names the files as follows, eg: "GP010021.MP4", I would like to be able to assign it a name chosen by me, for example, "H8-01". It's possible? Thanks!

#goprohero8 #namingfile #takevideo

KonradIT, 2021-09-13:
Use GoPro Labs, it will let you add a prefix. I also wrote an utility which helps you sort out the media files:

https://github.com/konradit/mmt

wilbercl, 2021-09-14:
Hello thanks for responding, I searched the internet and followed the steps to install GoPro Labs FW 01.70.75, and when creating the QR code with my prefix, the camera scans it but when taking a video it is not saved, that is, it does not work for me :-(

https://gopro.com/en/us/support/hero8-black-product-update

wilbercl, 2021-09-14:
It works, but I don't know why the recorded files are not visible inside the GoPro, but when you remove the microSD and put it in the computer, the files do come out.