KonradIT/goprowifihack #263: Set Linear in Photo Mode

shubham9436, 2021-12-07:
**Problem:**

How to set Linear lens in Photo Mode?
In constants.py, I can see FOV paramter, but there is no such parameter in Photo class

**Details:**

GoPro Camera(s): Hero 8
Firmware Version: 1.2

grcanepa, 2021-12-07:
Not using Phyton, but from what I can see working with Hero Max (which is very similar) "linear" lens mode can only be set when the camera is in Video/TimeWarp type mode not photo 

From https://github.com/KonradIT/goprowifihack/blob/master/MAX/MAX-Commands.md
Lens (Video/TimeWARP)
Max SuperView: http://10.5.5.9/gp/gpControl/setting/121/7
Wide: http://10.5.5.9/gp/gpControl/setting/121/0
Linear: http://10.5.5.9/gp/gpControl/setting/121/4
Narrow: http://10.5.5.9/gp/gpControl/setting/121/6
Dual 360: http://10.5.5.9/gp/gpControl/setting/121/5
Lens (Photo)
Max SuperView: http://10.5.5.9/gp/gpControl/setting/122/18
Wide: http://10.5.5.9/gp/gpControl/setting/122/15
Lens (TimeLapse)
Max SuperView: http://10.5.5.9/gp/gpControl/setting/123/18
Wide: http://10.5.5.9/gp/gpControl/setting/123/15

If you check the settings on your camera there should not be a "linear" lens option in "Photo" mode, just the video modes

you can change the camera mode Video <=> to Photo with the following calls

Video: http://10.5.5.9/gp/gpControl/command/set_mode?p=1000
Photo: http://10.5.5.9/gp/gpControl/command/set_mode?p=1001

Hope this helps

shubham9436, 2021-12-07:
Well, I have not worked with Hero Max. There might be few changes with new GoPros like setting of Lens in Photo Mode.
I have attached the image of the same.

[image]

shubham9436, 2021-12-07:
Ok.. I kind of figured it out... 
Happy to share with everyone :)
Hope you can merge these changes. 
 

http://10.5.5.9/gp/gpControl/setting/122/x"
x represents
0 - Wide
10 - Linear
17 - Narrow

closed 2021-12-07 by shubham9436