KonradIT/goprowifihack #276: Stream hangs after 24.75 seconds on HERO5?

Hauteknits, 2022-07-29:
**Problem:**
Using a GoPro Hero 5 black. When setting up a livestream using the instructions provided and bitrate set to 1.0 and 2.0 Mbps, the stream will hang at roughly 24.75 seconds
Describe your problem as accurately as possible, include key details that help resolve your issue.

**Details:**
Using latest ffmpeg release w/ command `ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://10.5.5.9:8554` or `ffplay -fflags nobuffer udp://10.5.5.9:8554`. Tested on both Win10 and Arch. First time attempting to livestream so very open to suggestions. Site in chrome is `http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`

* GoPro Camera(s): HERO5 Black
* Firmware Version: 02.70
* Steps to reproduce:
 * Connect to gopro using wifi
 * set the gopro to 2.0 or 1.0 Mbps
 * open `ffplay` using commands above
 * start stream via chrome
* Happens every time? [Y/N]: Y

KonradIT, 2022-07-29:
Make sure you're sending the Keep Alive packets, check out how it's done: http://github.com/konradit/goprostream

closed 2022-07-29 by KonradIT

Hauteknits, 2022-07-29:
I was not using keep alive packets lol. I ended up looking through your python script, saw the keepalive, and came back here to close the issue lol 😂. Thanks for the help man!