KonradIT/goprowifihack #283: GoPro stream latency on android app

Kanhoto, 2022-10-27:
**Problem:**
I made an android apk for displaying GoPro stream, but when I start it, I have 3 or 4 seconds of latencies.
But when I cange the framerate using the GoPro, it seems to reload the stream and the latency go back to 1 second or less.
So I tried using Wifi command to change framerate but it doesn't work.
I suppose the problem is coming from my settings because it keeps packets it shouldn't keep.
`http://10.5.5.9/gp/gpControl/setting/3/9`

**Details:**
To receive that stream, I'm using libvlc and FFmpeg-kit :

> ArrayList<String> _Options = new ArrayList<String>();
        _Options.add("--file-caching=2000");
        _Options.add("--network-caching=150");
        _Options.add("--clock-jitter=0");
        _Options.add("--live-caching=200");
        _Options.add("--clock-synchro=0");
        _Options.add("--drop-late-frames");
        _Options.add("--skip-frames");
        _Options.add("-vvv");
        _myLibVlc = new LibVLC(findViewById(android.R.id.content).getContext(), _Options);
        _myPlayer = new MediaPlayer(_myLibVlc);
        _myVout = _myPlayer.getVLCVout();

> private String cmd = "-an -flags low_delay -flags2 fast -fflags nobuffer -f:v mpegts -probesize 2048 -r 12 -i udp://10.5.5.100:8554 -max_delay 1000 -preset ultrafast -vcodec libx264 -tune zerolatency -f mpegts -vcodec copy udp://127.0.0.1:12345";

> FFmpegKit.execute(cmd);

* GoPro Camera(s):
GoPro HERO 5

* Firmware Version:
Android 12
libvlc : org.videolan.android:libvlc-all:3.1.12
FFmpeg-kit : libs/ffmpeg-kit-full-4.5.1-1.aar

* Steps to reproduce:
Launch GoPro stream with latency and change the video mode framerate to 24 (even if it's already 24)

* Happens every time? [Y/N]: Y

sepp89117, 2023-01-19:
Use Exoplayer with LoadControl and set minBuffer to 900ms and maxBuffer to 1100ms. For me it works with Hero5 and Hero8 without problems.
Example: [PreviewActivity.java](https://github.com/sepp89117/GoEasyPro_Android/blob/master/app/src/main/java/com/sepp89117/goeasypro_android/PreviewActivity.java#L134) 

And by the way: ffmpeg-kit-min is enough for that. This will make the APK file smaller.

Kanhoto, 2023-01-19:
Thanks for the tip

But I kind of figured out that my options weren't working because I was declaring them like every stack overflow answer.
Though, it works when I did it that way :

> 
        ArrayList<String> _Options = new ArrayList<>();
        _Options.add("--file-caching=2000");
        _Options.add("-vvv");
        _myLibVlc = new LibVLC(findViewById(android.R.id.content).getContext(), _Options);
        _myPlayer = new MediaPlayer(_myLibVlc);
        _myPlayer.getVLCVout();
        _myPlayer.attachViews(_myVideoView, null, false, false);
        Media media = new Media(_myLibVlc, Uri.parse("udp://@:12345"));
        media.setHWDecoderEnabled(true, true);
        media.addOption(":file-caching=0");
        media.addOption(":network-caching=0");
        media.addOption(":live-caching=0");
        media.addOption(":clock-jitter=0");
        media.addOption(":clock-synchro=0");
        media.addOption(":drop-late-frames");
        media.addOption(":skip-frames");
        _myPlayer.setMedia(media);
        //_myPlayer.setRate(1.5f);
        _myPlayer.play();

sepp89117, 2023-01-23:
That makes me happy! 
What latencies do you have now?
I have about a second latency with my solution. 

Best regards

Kanhoto, 2023-01-23:
I also have about a second (which is better than 3-4 seconds that were getting longer over time)
But if I try to set playback x2, it nearly catch up to be around 300ms, then it stop because the buffer is empty
(normal since I'm trying to play x2 with a live)