KonradIT/goprowifihack #288: Wifi streaming with GoPro Hero 3+ steady problems / disconnects

will-code-for-pizza, 2023-03-19:
**Problem:**

After receiving a live stream via wifi, connection disconnects after 1-2 seconds

**Details:**

* GoPro Camera(s): Hero 3+ Black
* Firmware Version: HD3.11-firmware.bin
* Steps to reproduce:

- connecting Linux computer to GoPro Wifi (success)
- get IP address (10.5.5.105) from GroPro (10.5.5.9)
- ping 10.5.5.9 succeeded over long time
- as decribed here https://github.com/KonradIT/goprowifihack/blob/master/HERO3/Livestreaming.md

1. open http://10.5.5.9/camera/PV?t=MyGoProWifiPassword&p=%02 via browser or 'curl -L'
2. open http://10.5.5.9:8080/live/amba.m3u8 with VLC Player (stream connects for 1-2 seconds, then disconnects)
3. do a 'curl -L http://10.5.5.9:8080/live/amba.m3u8' and receive the following output
```
#EXTM3U
#EXT-X-TARGETDURATION:2
#EXT-X-VERSION:3
#EXT-X-ALLOW-CACHE:NO
#EXT-X-MEDIA-SEQUENCE:923
#EXTINF:1.00000,
amba_hls-3.ts
#EXTINF:1.00000,
amba_hls-4.ts
#EXTINF:1.00000,
amba_hls-5.ts
#EXTINF:1.00000,
amba_hls-6.ts
#EXTINF:1.00000,
amba_hls-7.ts
#EXTINF:1.00000,
amba_hls-8.ts
#EXTINF:1.00000,
amba_hls-9.ts
#EXTINF:1.00000,
amba_hls-10.ts
```
4. try to connect to a specific stream URL, f.e. http://10.5.5.9:8080/live/amba._hls-8.ts

* Happens every time? [Y/N]: Y

will-code-for-pizza, 2023-03-19:
Update:
If I set VLC player to "loop" mode, the stream reconnects every 5 seconds and plays for 1-2 seconds in a very bad quality.
So, no solution yet.

will-code-for-pizza, 2023-03-19:
There is another m3u file. Connecting web browser to http://10.5.5.9:8080/live/ is shows

```
[Name](http://10.5.5.9:8080/live/?order=N)	[Size](http://10.5.5.9:8080/live/?order=s)
[   ]	[aaba.m3u8](http://10.5.5.9:8080/live/aaba.m3u8)	352          <<-------   !!!!
[   ]	[amba.mp4](http://10.5.5.9:8080/live/amba.mp4)	link
[   ]	[amba.m3u8](http://10.5.5.9:8080/live/amba.m3u8)	349
[   ]	[amba_hls-1.ts](http://10.5.5.9:8080/live/amba_hls-1.ts)	37K
[   ]	[amba_hls-2.ts](http://10.5.5.9:8080/live/amba_hls-2.ts)	37K
[   ]	[amba_hls-3.ts](http://10.5.5.9:8080/live/amba_hls-3.ts)	36K
[   ]	[amba_hls-4.ts](http://10.5.5.9:8080/live/amba_hls-4.ts)	36K
[   ]	[amba_hls-5.ts](http://10.5.5.9:8080/live/amba_hls-5.ts)	36K
[   ]	[amba_hls-6.ts](http://10.5.5.9:8080/live/amba_hls-6.ts)	0
[   ]	[amba_hls-7.ts](http://10.5.5.9:8080/live/amba_hls-7.ts)	37K
[   ]	[amba_hls-8.ts](http://10.5.5.9:8080/live/amba_hls-8.ts)	38K
[   ]	[amba_hls-9.ts](http://10.5.5.9:8080/live/amba_hls-9.ts)	36K
[   ]	[amba_hls-10.ts](http://10.5.5.9:8080/live/amba_hls-10.ts)	37K
[   ]	[amba_hls-11.ts](http://10.5.5.9:8080/live/amba_hls-11.ts)	36K
[   ]	[amba_hls-12.ts](http://10.5.5.9:8080/live/amba_hls-12.ts)	36K
[   ]	[amba_hls-13.ts](http://10.5.5.9:8080/live/amba_hls-13.ts)	36K
[   ]	[amba_hls-14.ts](http://10.5.5.9:8080/live/amba_hls-14.ts)	37K
[   ]	[amba_hls-15.ts](http://10.5.5.9:8080/live/amba_hls-15.ts)	36K
[   ]	[amba_hls-16.ts](http://10.5.5.9:8080/live/amba_hls-16.ts)	36K
```

KonradIT, 2023-03-19:
Try using MPV to open the link `http://10.5.5.9:8080/live/amba.m3u`.